skiselev/micro_8088 #1: FE2010 Datasheet

crasbe, 2017-11-06:
Hey,

I don't know if you have found this yet, but here is the datasheet of the FE2010 from Faraday: https://www.tvsat.com.pl/pdf/f/fe2010_far.pdf (also attached).

This might give some insightes and also features a reference design.

Greetings,
Chris
[fe2010_far.pdf](https://github.com/skiselev/micro_8088/files/1446091/fe2010_far.pdf)

skiselev, 2017-11-06:
Hi Chris,

Thank you for contacting me!

I do have this datasheet. I probably should put it in the GitHub repository, together with the datasheets of other components used in this project.

I am looking for FE2010A datasheet. The closest I have is Intel Wildcard 88 datasheet. That module is apparently based on FE2010/FE2010A, and the datasheet documents some chipset registers. I also have schematic for Commodore PC10-III, which is also based on FE2010A.

Thanks,
Sergey

skiselev, 2017-11-25:
I've added the Faraday FE2010 datasheet into the repository - Documentation directory.

closed 2017-11-25 by skiselev