skiselev/micro_8088 #4: Tie pins 1 and 30 of U4 to ground in next revision?

KyleJ61782, 2021-03-08:
So I received a pre-built Micro 8088 except they put in an SST39SF040. Practically that's not really an issue because it appears they flashed 4 images onto it at 128K intervals, but the problem is that the BIOS writes settings back to flash which could be an issue because pins 30 (A17) and 1 (A18) are floating, so their values in theory could change. See the pin assignments from the data sheet:

[image]

I've already changed the BIOS settings write code to handle the larger flash, but to ensure the address lines aren't floating I'm just going to put in bodge wires tying them to ground, so it won't be a problem. Perhaps, though, connecting both pins to ground in the next revision opens up more substitution possibilities?

skiselev, 2021-03-08:
It makes sense. It also should support 29F020/39SF020 ICs in this case. By the way, do you mind contributing the flash programming code changes to the https://github.com/skiselev/8088_bios project?

KyleJ61782, 2021-03-08:
Absolutely! I'm working on a few things related to the BIOS at the moment and plan on submitting all of the code back. I'm working on the uflash code at the moment, enabling that to work with bigger chips as well.

prabathbr, 2023-06-26:
> So I received a pre-built Micro 8088 except they put in an SST39SF040. Practically that's not really an issue because it appears they flashed 4 images onto it at 128K intervals, but the problem is [...]

> It makes sense. It also should support 29F020/39SF020 ICs in this case. By the way, do you mind contributing the flash programming code changes to the https://github.com/skiselev/8088_bios project?

Hi, I ordered a "SST39SF020A-70-4C-PHE-ND" instead of SST39SF010A as it is backorder in Digikey. I though I can just ground PIN30 and use it. Would I need to do any other changes to the BIOS code ?

skiselev, 2023-11-08:
> Hi, I ordered a "SST39SF020A-70-4C-PHE-ND" instead of SST39SF010A as it is backorder in Digikey. I thought that I can just ground PIN30 and use it. Would I need to do any other changes to the BIOS code ?

The latest 8088 BIOS version supports SST39SF020A and SST39SF040 ICs