skiselev/micro_8088 #5: Proton PT8010AF instead of Faraday FE2010A ?

explit7, 2021-11-07:
Hello Sergey.
This guy wrote here, that Proton PT8010AF is a clone of Faraday FE2010A.
http://dosdays.co.uk/topics/faraday_chipset.php

While its very difficult to find Faraday FE2010A. and the price tags are very high,
the Proton PT8010AF is cheap and can be purchased for adequate price on aliexpress:

https://www.aliexpress.com/item/1005002678888959.html?spm=a2g0o.productlist.0.0.1677a464jQmYvE&algo_pvid=9bbdc611-5651-4608-87a1-304ddd56c033&algo_exp_id=9bbdc611-5651-4608-87a1-304ddd56c033-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000021678893192%22%7D

Dp you have some information about this chip, could it be used as a direct replacement, or are software changes (eg. BIOS) needed?

Thanks
[image]
[image]

skiselev, 2021-11-08:
The information that Proton PT8010AF is possibly a clone of Faraday FE2010A is actually coming from me... But that's a good point. I have a motherboard based on Proton PT8010AF, and I can try to check the pinout and confirm if it is the same as Faraday FE2010A, also perhaps poke chipset registers and see if they are the same as FE2010A... Maybe later this week

skiselev, 2021-11-08:
I did a bit of reverse engineering on my Proton PT8010AF board, and I can confirm that the pinout is the same as that of Faraday FE2010A.
I haven't done any software testing yet. There might be some differences there. I've ordered some Proton PT8010AF ICs, and I'll test Micro 8088 when they arrive.

explit7, 2021-11-08:
Really cool, thanks, will also order one to test!

skiselev, 2021-11-18:
I received a Proton PT8010AF IC, and tested it on the Micro 8088 board.
It does work. Although, it appears that it inserts more wait states than Faraday FE2010A does, and as the result the system runs slightly slower. Not a significant difference, also when running on 9.55 MHz, it might be possible to reduce the number of the wait cycles by setting both bits 7 and 6 of the chip configuration register at port 0x63, which brings the performance to the same as FE2010A performance with only bit 7 set - the default turbo mode in my BIOS.

explit7, 2021-11-18:
thanks for the info, Sergey! this are indeed great news! That means for many - the search for the FE2010A is over, and PT8010AF could be bought for adequate precises!

explit7, 2021-11-18:
Should it means, that you need to change something in BIOS ?

skiselev, 2021-11-18:
It will work as is, just a bit slower than FE2010A. I am still checking if there is a way to detect PT8010A and adjust the configuration...

skiselev, 2021-11-21:
It appears that the slowness of PT8010AF vs. FE2010A is caused by DRAM refresh being enabled by default in PT8010AF, and disabled in FE2010A. I am not 100% certain what causes this difference. The BIOS does enable refresh on both.
I am planning to update the BIOS not to enable DRAM refresh. It is not needed anyway since Micro 8088 and related projects use SRAM instead of DRAM

eluancm, 2021-11-23:
Just FYI, utsource cancelled my order and asked for 3x the initial price, so it looks like this is going the FE2010A way fast.

skiselev, 2021-11-23:
@eluancm same happened to me.
It is quite possible that Utsource people (or their suppliers, since I don't think Utsource really stocks anything) read this or Twitter posts, or simply observe the demand, and adjust prices. They had some ICs laying around without anyone asking for them for 30 years, and suddenly the demand picked up. Well, another incentive for me to develop systems that don't rely on old/proprietary components.

skiselev, 2021-11-23:
The answer is "yes", Proton PT8010AF can be used instead of Faraday FE2010A, if you're lucky to find it.
Closing this ticket...

closed 2021-11-23 by skiselev

wrljet, 2021-11-23:
Mine says "departed shipping partner".  Ordered on 11/18.

I hope they aren't counterfeit/fake.

Bill

explit7, 2021-11-23:
mine came last week. got it for 17,- eur including shipping.

wrljet, 2021-12-01:
Mine arrived 30 Nov.  Ordered on 18 Nov.