skiselev/micro_8088 #9: Missing a reset pull-up resistor

andykarpov, 2022-07-03:
FE2010A datasheet says that the reset line should be equipped with a 10k pull-up resistor, that is missing on the current design.
Adding that 10k pull-up increased stability of the board's power-up and reset.

P.S. Also i've added a 47k resistor in parallel of the 10uF reset capacitor to discharge it after power-off, so it can produce a proper initial reset signal for the next power-on cycle.

skiselev, 2022-07-03:
/RESIN has a pull-up resistor (RR2).
I don't think CLEAR/RESET should have one.
It could be a good idea to connect a diode between /RESIN and VCC to discharge it when system is powered down

andykarpov, 2022-07-03:
Ah, yes, thanks, i see, and sorry about the invalid issue then :)
Regarding the discharging diode - well, nice idea, let's try!

closed 2023-02-01 by skiselev