skiselev/micro_8088 #12: Can't write Flash, nor FDD

TzOk83, 2022-12-28:
After some minor issues, I've managed to start Micro8088, but I have 2 problems:
1. I can't store BIOS Setup settings in Flash.
2. I can't write anything to FDD.

ad. 1. When I try to save settings in BIOS setup, I got ERROR, and it continues to boot. I have SST 39SF020 Flash chip on board, with pins 1 and 30 tied to GND.

ad. 2. When I try to write anything to floppy (to be more accurate to GOTEK/Flash Floppy FDD emulator) the computer hangs. I've tried two different FDD controllers (one I'm using now is a PTI227W Multi I/O 16-bit ISA card based on W83757F+W83758P).

Don't know if this has anything to do with the issue, but the computer is hard to boot. After powering on, it often freezes, with no activity on the bus. The other small issue was, my 28.6363 MHz crystal was oscillating at 1/3 of that frequency - the computer was running, but very slow. Now I've put 14.3 MHz taken out of some scrapped modern motherboard (from the LPC bus), and it runs fine.

When I try to run BASICA from PC-DOS 3.30 I get "Divide overflow" error.

I was unable to get the recommended Faraday chipset, so I'm using a Proton clone. CPU is NEC V20 (I also have tried the original i8088), and all "TTL" chips are 74HCT series (it is really hard to get anything else).

Except for those issues, the computer seems to run fine and stable.

What I can/should check to resolve that issues? I have a simple logic and a 4 channel DSO.

skiselev, 2022-12-28:
Your issue documents multiple problems. Next time please submit separate issues. It will make it easier for me to address, and for others, who encounter similar issues, to find the discussion.

1. Two possible issues here: (1) You are not using U16 SPLD, instead you use U14 74*04 logic IC. In this case, the system will not support writing to the Flash ROM. (2) Support for SST39SF020 was just recently added to 8088 BIOS, and it is not a part of the image published in this GitHub repo. Please clone or download the source from [Micro 8088 GitHub repo ](https://github.com/skiselev/8088_bios) 
2. Multiple things here again. (A) It is still not clear to me what frequency you're using. The FE2010A can work with either 28.63636 MHz or 14.31818 MHz crystals. I am not sure if the Proton PT8010AF supports 14.31818 MHz frequency. In any case, if using 14.31818 MHz crystal, make sure that JP2 XSEL jumper is open. Wrong (too low) frequency might be one of the reasons the FDC does not work properly, or create stability issues. Measure the actual CPU frequency (CLK signal on the CPU). Ultimately, I recommend getting the 28.63636 MHz crystal documented in the BOM, or a crystal with very similar characteristics (B) BASICA will not work on Micro 8088, or for that matter on any IBM PC clone, because it requires built-in ROM BASIC, which was only provided with the original IBM computers. Use GWBASIC instead (C) I do not recommend using 74HCT series, they do not provide enough current to drive ISA bus. Use 74F series, or if you cannot find them, try 74LS or 74ALS (they are not guaranteed to work, but better chances than with 74HCT).

TzOk83, 2022-12-29:
Sorry for 2 issues in one post, but I believed, they are connected with each other.

ad. 1. I'm using SPLD. I thought it was enough to hold additional address lines low, but it is apparently not, so I'll try to compile BIOS from source. 

ad. 2. I'm using 14.3 MHz crystal and JP2 is open. Both clocks present on ISA bus have correct frequencies and look ok. 74HCT have way higher output current capabilities than LS (25 mA vs 8 mA), yet they are slightly slower (22 ns vs 15 ns). I'm trying to locally source either LS or F series chips, but without any luck. Reference design from Faraday is using LS chips as well as my ISA cards do.

TzOk83, 2022-12-29:
I can close this issue, as compiling BIOS from the most recent source solved writing settings to Flash, and FDD issues were completely my fault, unrelated to the hardware.

closed 2022-12-29 by TzOk83