skiselev/micro_8088 #15: Strange Problems with Proton PT810AF

explit7, 2024-01-31:
Hello collegues.
I have a great, fully working Micro8088 Version 1.3 with NEC V20 and Faraday / WD FE2010A.

Now i have build another Micro8088 Board (Version 1.1). My Problem is, that both boards works without any problems with the FE2010A but if i put PT810AF instead - i see the BIOS screen, and the board immediately resets.
Sometimes the reset is very quick, sometimes, i see full BIOS screen.

This is the second Proton PT810AF which i bought at Aliexpress from China. From 2 different sellers. And both chips show the same problems which are not there with Faraday FE2010A.

Maybe its a defective stock of Proton PT810AF Chips?

[image]
[image]
[image]

skiselev, 2024-01-31:
One of my Micro 8088 currently has the Proton PT8010AF controller. It works just fine. That one particularly says "JAPAN" on it some place.
I have a few more Proton PT8010AF from a different batch, with a different marking. I don't think I tested them yet, I'll give them a try.

In your case, are these two PT8010AF from different sellers parked similarly? Maybe you can do a side to side photo from top and the bottom?
Another idea: check the "Fast wait states" setting. Maybe disabling wait states changes the behavior?

explit7, 2024-01-31:
Thanks for the info, Sergey, The 2x PT810AF has the same marking and was packed similar. So i think 2 different sellers shipped the PT810AF from the same batch. I flashed a new BIOS, so the frequency settings and wait state settings are default. Will try to put FE2010A and disable wait states manually (just to be sure).

At the moment i think that the full batch is somehow defective.

skiselev, 2024-02-01:
I tested PT8010AF from the other batch I have, and it works just as well as one that I had in the system.
The one which I normally have in the system is in the middle, and the new one is on the right.

Maybe there is something else going on? For example, power supply is not stable, which somehow affects the PT8010AF more than FE2010A?

Here are the photos for your reference:
[image]
[image]

explit7, 2024-02-01:
My Proton looks the same way like your on the right side. All AliExpress PT8210AF look this way.
The Proton in the middle seems to be an original Japan-made one.

I am very curious, because all who has Micro8088 Boards here in Germany, bought Proton PT8210AF in China on AliExpress, and they works just fine.

I think the whole batch of this chips may be defective, because with WD FE201A the same board is working absolutely fine.

The last thing, what i want to test is 8088 CPU instead NEC V20.

This is my other Micro8088 Mainboard Version 1.3 which is also working fine with FE201A
[image]