skiselev/micro_8088 #18: clock speed setup by keyboard doesn't work

giobbino, 2024-02-21:
Hi,
not a big issue, but...

the Ctrl-Alt-Keypad -/+/* sequence doesn't work (at least for me). I checked the speed with the speed.com program and there is no difference at all.  

Just to be sure, I tried to modify the clock speed by the Bios, and the speed.com routine correctly shows the different frequency.

skiselev, 2024-02-21:
What platform are you using?
It will not work on Book8088 because turbo there is not software controllable

giobbino, 2024-02-21:
micro_8088

giobbino, 2024-02-21:
I suspected it, so I already did that test before to open the issue, trying to modify the speed from the dos prompt. It doesn't change. I tried many times, i.e. boot (BIOS says 9.45 MHz) -> prompt ms-dos -> press Ctrl/Alt/-  -> speed.com -> nothing changes.

Just a question: should this function modify the BIOS value too? Because I did that test too, booting and then trying to modify the speed from dos prompt, by keyboard, and then reboot. No changes at all.

I also suspected the keyboard, so I tried with another one, different model. Nope.

skiselev, 2024-02-22:
I [tested it and it works for me](https://youtu.be/vTrLP7aXUKg?si=SCVkONw-54LqYj1T).
Maybe it has to do with the "Pass E0 code" switch (SW2.3)? What's the current setting of that switch in your system?

giobbino, 2024-02-22:
It was OFF, but I tried to put it to ON, no changes. Just to be sure, I downloaded the same tool you're using, systest, and tried both switch modes.  The single keys off the keyboard (ctrl, alt, keypad +, - , *) are working.

BTW I believe it doesn't any difference, but today I discovered I did a mistake while mounting the backplane, I put all caps 10 uF, but I've seen the C20 is 1 uF.  If you think it could make any difference, I'm assembling a 2nd micro_8088, this time with the right cap. It will take a day or two yet...