skiselev/micro_8088 #19: 3D print full desktop cabinet for 8088 oh thingiverse

giobbino, 2024-03-23:
Hello,
I released a project for a full desktop cabinet (CPU, PSU, Floppy disk drive) for the micro_8088 computer. I made it suitable for the Creality Ender 3 Pro, so it should be ok for most printers (it requires a printer with at least 220x220 mm print size).

https://www.thingiverse.com/thing:6546330

There are two flavors, a smaller one that will fit perfectly the micro_8088 and all the Sergey's cards, and a bigger one that can also host the LPT + COM card by hkzlab.

Hope this would help!

- Giovi

lpsantil, 2024-03-23:
You might also be interested in this AT/386/486 style case.

https://www.vogons.org/viewtopic.php?f=46&t=85634&start=20

CodeAsm, 2024-03-27:
> You might also be interested in this AT/386/486 style case.
> 
> https://www.vogons.org/viewtopic.php?f=46&t=85634&start=20

So much plastic, but that vogons thread case looks awesome. maybe a metal behind and im happy. both cool projects tho, cause not everyone can bend metal or dares to touch sharp metal

giobbino, 2024-03-27:
Indeed, the 386/486 case is great... much, much better than mine. Unfortunately you need at least a 300x300 mm bed, so you can't print it with an entry-level printer like the Ender 3 and so on. Maybe when I will change my Ender 3 with a bigger one - I'm also waiting for the Sergey's 386 project  ;-)

I'm not actually concerned about how much filament I use, it's quite cheap _(especially if compared with the micro_8088 budget - all the PCB cards and backplane, the needed components, the PSU, etc.)_ ; so I decided to make a strong case. Using a 10-20% infill, you can print everything with less than a filament spool.  

Also, for an XT-clone, I prefer the desktop model, with a monitor on top of it (I'm using an 8" LCD display, 4:3, I bought on Aliexpress with a 3D-printed case), instead the tower model.