skiselev/micro_8088 #22: Is there any chance for a similar mobo supporting 286/386 CPUs?

pshirshov, 2024-08-07:
Just re-posting my question from NuXT issue tracker: https://github.com/monotech/NuXT/issues/13

I would love to see a similar motherboard supporting 286/386 CPUs. Is there any chance for that?

explit7, 2024-12-08:
IMHO there will be a problem to fine Chipset Chips for 286/386. The goal of Micro8088 is that it based on a single FE2010(A) or 8210 chip. As i know, there are now 286/386 single chips Mainboard which can be easy soldered. There exist very small 386 Mainboard based on a single chip, but its very difficult to find this chip and to solder it.

pshirshov, 2025-01-11:
I'm surprised it might be the case, I believe that 386 had extremely wide adoption in various industrial applications and was manufactured until recently. It's definitely the case for 286, I'm wondering if a cheap FPGA might be used to replicate the relevant chips.