skiselev/micro_8088 #23: Pc works but when load dos no scroll on c: no on screen what im typing if i type dir commands wirk but cant see what i erite on screen if run fdisk and escape cursor turns to normal

subzero2379, 2024-11-12:


skiselev, 2024-11-12:
A bit difficult to understand what's going on there. It seems like you're having display issues.
It would be nice to get a screenshot, a photo, or a video showing the issue, as well as a full description of your configuration, for example the type of the display/graphics card, type of the display.
Also, as troubleshooting steps, you can try to isolate the issue... Does it still happen if you use another display. Does your graphics card and display work with another system?

Generally speaking, the Micro 8088 and 8088 BIOS doesn't have much to do with the display, unless it is an MDA or a CGA display. In case of a EGA or a VGA/SVGA displays, the graphics card has its own BIOS extension.

It is also possible that newer LCD displays do not properly support some older VGA modes, and you are not seeing the top or the bottom of the image...

subzero2379, 2024-11-13:
you are right its a vga bios problemm 
i will try to burn a later rom and see what happens